CrossTheRoadElec/Phoenix-Releases #1: Phoenix Libraries are cleared on roboRIO reboot.

Pauldg7, 2019-01-10:
Problem:
Phoenix Libraries are removed on reboot of roboRIO.

Steps to reproduce:
1) Installed Phoenix Library/Diagnostics
2) Install Success
3) Reboot roboRIO
4) ERROR 13 Call Library Function Node in CTRE_Phoenix_TalonSRX_Open.vi:3800001->Begin.vi->Robot Main.viLabVIEW: Failed to load dynamic library because of missing external symbols or dependencies, or because of an invalid file format.

What I've tried: 
I've been fixing it by just running the install again and restarting robot code. I've also tried the experimental deployment software and it still doesn't save the libraries on restart.

Thanks! Ri3D team ZouKeepers.

ozrien, 2019-01-10:
#### Questions
Did you run our installer on your dev PC...
https://phoenix-documentation.readthedocs.io/en/latest/ch05_PrepWorkstation.html#option-1-windows-installer-strongly-recommended
... then opened Tuner from the start menu afterwards?  

Or did you grab the GitHub tuner? If so, how did you install the CTRE Phoenix LabVIEW VIs?

#### Possible  Root-causes
- our library in the RIO has been removed or modified after restart.
- RIO was re-imaged
- VIs on the deploying PC are from last season or don't match our tested release (check installer list in windows)
- The NI LabVIEW deploy is doing something odd.
- something odd in the RIO filesystem

#### Next Steps
If you are ssh-savy, confirm ```libCTRE_PhoenixCCI.so``` still exists by ssh-ing into RIO...
```/usr/local/frc/third-party/lib``` ... and post results.  
Example: "ls -l /usr/local/frc/third-party/lib"
Do this after your have reproduced the problem.

If you are not, contact support@ctr-electronics.com, and we will walk you through this.
You should probably do this regardless of your findings.

Pauldg7, 2019-01-10:
- I've only had to image the roboRIO to the 2019 image once the first day of the season Jan 5th.
- I don't have any previous FRC installs on this PC as this is a new laptop a few weeks old.
- I have the latest version of the CTRE TalonSRX Labview library from the CTRE Phoenix installer. I also know it is the latest because one of my team mates reported a bug in the Phoenix Labview repo for which we were informed it was patched on master and hasn't been released yet.

Installer:
CTRE Phoenix Framework v5.12.0.1 ran on dev pc
Running "ls -l /usr/local/frc/third-party/lib"

While having ERROR 13 as stated above:
-rw-rw-r--    1 lvuser   ni             337 Jan  9 22:17 ReadMe.txt
-rwxr-xr-x    1 lvuser   administ    400196 Jan 10 16:30 libCTRE_PhoenixCCI.so*

After running Install Phoenix Library/Diagonostics:
-rw-rw-r--    1 lvuser   ni             337 Jan  9 22:17 ReadMe.txt
-rwxr-xr-x    1 lvuser   administ    400196 Jan 10 17:32 libCTRE_PhoenixCCI.so*

I can explore around the file system some more and come back to you with what I find but the file seems to exist on the rio when I get the issue.

ozrien, 2019-01-10:
I have two thoughts...
- "chmod 777 libCTRE_PhoenixCCI.so" and rerun experiment.
- remove libCTRE_PhoenixCCI.so in the RIO file system, reboot, cleanly re-update via Tuner, and rerun experiment.

Pauldg7, 2019-01-11:
Ran both, still error 13 on reboot of roboRIO. Going to try and borrow a roboRIO from another robot and try to reproduce.

ozrien, 2019-01-11:
I have a hypothesis.  Can you check /usr/local/frc/lib and see if there is a libCTRE_PhoenixCCI.so there?
I suspect an older or incorrect version of the lib could have made it to '/usr/local/frc/lib' which is the old location.  Running LifeBoat on accident, for example may do this.

So here's the experiment,
$ rm /usr/local/frc/lib/libCTRE_PhoenixCCI.so
$ rm /usr/local/frc/third-party/lib/libCTRE_PhoenixCCI.so
$ sync
$ reboot now

When RIO comes back up, run Tuner to cleanly install (note it installs into third-party).
Then run the negative test (reproduce erroneous behavior).

ozrien, 2019-01-12:
We were able to reproduce this symptom by using last year's Hero LifeBoat to deploy 2018 libs on our 2019 RIO, then using Tuner to install libs.  When Tuner installs the RIO, LV works.  Then power cycle, and LV appears to try to load the old libs in the old location, causing the original symptom.

- If you perform $ rm /usr/local/frc/lib/libCTRE_PhoenixCCI.so in shell, it will resolve the issue.
- Or delete the legacy file using your favorite method.
- Or image RIO, and use Tuner to again install library (not LifeBoat).  
Note that imaging the RIO really isn't necessary, and its super slow, so just delete the legacy file.

Moving forward, Tuner will auto delete this for you, and future versions of Lifeboat won't allow you to install 2018 bins on a 2019 RIO.

closed 2019-01-14 by ozrien

Pauldg7, 2019-01-14:
@ozrien Thanks! I can confirm that this fix works.