CrossTheRoadElec/Phoenix-Releases #5: Phoenix Tuner 1.5.10 not available to download separately

Sanjana679, 2020-02-03:
Phoenix Diagnostics Tuner 1.5.9 is available for download, but Phoenix Tuner 1.5.10 is not.

JCaporuscio, 2020-02-03:
Phoenix Tuner is usually provided separately when it's released earlier than the Installer releases.
1.5.10 isn't up because it was released at the same time as the 5.17.6 installer release, and you can get it by running the installer (which you need anyway for new firmware files/library files).

Is there a specific reason you need the separate download, or is it just wanted as a nice-to-have?

Sanjana679, 2020-02-08:
I originally thought that it would be possible to download the phoenix
tuner separately so it could be accessible on a mac or linux machine.
However, I have realized that it is not possible for that to happen.

*Sent with Shift
<https://tryshift.com/?utm_source=SentWithShift&utm_campaign=Sent%20with%20Shift%20Signature&utm_medium=Email%20Signature&utm_content=General%20Email%20Group>*

closed 2020-02-14 by JCaporuscio