CrossTheRoadElec/Phoenix-Releases #8: Publisher Stated As Unknown By Windows When Downloading

silverlightning926, 2020-10-19:
When downloading, multiple warnings pop up because the publisher of the project is not set. Fixing this would make it much easier to download and would seem much safer to people first getting into FRC. Currently, you have to go to the bottom of the warning panel to ask Windows to trust the software before you download it. This is normally hard to find.

JCaporuscio, 2020-10-19:
To clarify, do you mean you get warnings when downloading the zip file from GitHub or when installing the exe after downloading?

The warnings when installing are a known issue and we document them during the install process:
https://phoenix-documentation.readthedocs.io/en/latest/ch05_PrepWorkstation.html#option-1-windows-offline-installer-c-java-labview-hero-c
Fixing this requires signing the release binaries which we may do in the future (but I have no timeline for this).

Warnings on *download* would be a new issue we haven't seen before.  We have yet to see this issue for the common browsers (Chrome/Firefox/Edge).  Can you provide steps to reproduce and/or screenshots?

silverlightning926, 2020-10-19:
Ya, I was talking about the warnings after downloading and installing. No worries, as there is currently a workaround so it doesn't seem to be blocking anyone. Just wanted to let you know.

JCaporuscio, 2020-10-19:
Gotcha, thanks for clarifying.

I'm going to leave this open for now since we haven't decided on a path forward yet, and so anyone that sees it can reference the current documented workaround.

JCaporuscio, 2021-01-13:
Closing since this is documented in our read-the-docs

closed 2021-01-13 by JCaporuscio