CrossTheRoadElec/Phoenix-Releases #19: No setInverted(InvertType) method definition in IMotorController interface

richiksc, 2021-09-22:
Our codebase makes heavy use of the `IMotorControllerEnhanced` interface, as we have a custom implementation of the interface in order to virtualize motor controllers alongside the CTRE-provided implementations such as `TalonSRX` and `TalonFX`. The `setInverted(InvertType)` API is especially useful to us, as it allows us to set `InvertType.FollowMaster`.  However, we aren't able to make this call directly on `IMotorControllerEnhanced` references, since this method seems to be not defined in the interface - only `BaseTalon`. This might be an oversight, as `setInverted(boolean)` is available in the interface.

JCaporuscio, 2021-09-22:
Yep, this looks like an oversight.  We've now tracked it internally and we'll get it fixed in a future release.

CoryNessCTR, 2022-01-07:
This has been added in our 5.20.2.2 release

closed 2022-01-07 by CoryNessCTR