CrossTheRoadElec/Phoenix-Releases #20: Travis-ci build of code using Phoenix.json fails

Crossle86, 2021-09-28:
I have a project using phoenix.json (5.14.9) that I compile under Eclipse (Java) on  Windows PC and also on Travis-ci.com for deployment from github. This has worked fine for some time. Recently I updated the project to gradle 6.0.1 on advice of Wpilib team to get my project on lastest Wpilib release. Builds fine under Eclipse. However, when it  builds under Travis-ci, the build fails as follows:

Execution failed for task ':extractTestJNI'.
> Could not resolve all files for configuration ':nativeDesktopZip'.
   > Could not resolve com.ctre.phoenix.sim:cci-sim:5.19.4.
     Required by:
         project :
      > Could not resolve com.ctre.phoenix.sim:cci-sim:5.19.4.
         > Could not get resource 'https://devsite.ctr-electronics.com/maven/release/com/ctre/phoenix/sim/cci-sim/5.19.4/cci-sim-5.19.4.pom'.
            > Could not GET 'https://devsite.ctr-electronics.com/maven/release/com/ctre/phoenix/sim/cci-sim/5.19.4/cci-sim-5.19.4.pom'.
               > peer not authenticated
   > Could not resolve com.ctre.phoenix.sim:simTalonSRX:5.19.4.
     Required by:
         project :
      > Could not resolve com.ctre.phoenix.sim:simTalonSRX:5.19.4.
         > Could not get resource 'https://devsite.ctr-electronics.com/maven/release/com/ctre/phoenix/sim/simTalonSRX/5.19.4/simTalonSRX-5.19.4.pom'.
            > Could not GET 'https://devsite.ctr-electronics.com/maven/release/com/ctre/phoenix/sim/simTalonSRX/5.19.4/simTalonSRX-5.19.4.pom'.
               > No PSK available. Unable to resume.

I have not made much progress with travis-ci support so I thought I would troll this here in the off chance this is something known to ctre. It may be some sort of SSL credential or protocol issue but I haven't been able to pin it down. There is a lot of plumbing at play here that I am not that conversant with.

JCaporuscio, 2021-09-28:
As far as we're aware there's no issue with the maven host for Phoenix.
In fact all 4 of the links in your log are valid, SSL works and they can be followed directly to display the contents of the files referenced.

Is your project public?  I probably won't be able to do a deep dive but I can take a quick look.

Crossle86, 2021-09-28:
Thanks for whatever you can do. https://github.com/ORF-4450/RobotLib

JCaporuscio, 2021-09-28:
Nothing sticks out to me as obviously wrong.  I did a quick clone and build (`./gradlew build`) and it seems to build fine.
So I suspect something is unique about your travis setup that's causing the issue.

Crossle86, 2021-09-28:
Ok Thanks. Still waiting on travis support to look at bunch of data I collected so we will see. May just convert to github actions.

closed 2022-01-03 by CoryNessCTR