CrossTheRoadElec/Phoenix-Releases #21: 100gb of log files

allengregoryiv, 2021-12-25:
This was very possibly something that I did wrong but figured you'd all like to know this is possible.

Phoneix Tuner generated 100GB of log files. I haven't been able to open them in anything to see what's actually in the files. It lists a last modified date of May 19th, this is my personal computer so it's used regularly and I don't believe I left it running for weeks at a time or anything like that. 

[image]
.

ozrien, 2021-12-25:
According to our private trackers, this was fixed in 5.19.2 (Jan 2021).  I wonder if the Tuner version is too old.

Basically its possible for log files to grow large if Tuner is left open for an excessive period of time (weeks, months, or more).
Latest Tuner actually deletes these files if they are more than 7 days old on startup.  I recommend confirming what version of Tuner is installed.

allengregoryiv, 2021-12-25:
Okay thank you, I'll let you know if I see it again on an updated Tuner.

closed 2023-08-01 by JCaporuscio