CrossTheRoadElec/Phoenix-Releases #22: Talon FX Remote Reverse Limit Switch Device ID Not Configurable

dkt01, 2021-12-29:
When using `ConfigAllSettings()`, `reverseLimitSwitchDeviceID` is not applied properly.  It always defaults back to 0.

Example configuration sent:
```
.supplyCurrLimit = Current Limiting will activate if SUPPLY current exceeds 30 amps for 0.1 seconds.  Then current will hold at 30 amps;
.statorCurrLimit = Limiting is disabled.;
.motorCommutation = 0;
.absoluteSensorRange = 0;
.integratedSensorOffsetDegrees = 0.000000;
.initializationStrategy = 0;
.primaryPID.selectedFeedbackSensor = RemoteSensor0;
.primaryPID.selectedFeedbackCoefficient = 1.000000;
.auxiliaryPID.selectedFeedbackSensor = Integrated Sensor;
.auxiliaryPID.selectedFeedbackCoefficient = 1.000000;
.forwardLimitSwitchSource = LimitSwitchSource_RemoteTalonSRX;
.reverseLimitSwitchSource = LimitSwitchSource_RemoteTalonSRX;
.forwardLimitSwitchDeviceID = 4;
.reverseLimitSwitchDeviceID = 4;
.forwardLimitSwitchNormal = LimitSwitchNormal_NormallyClosed;
.reverseLimitSwitchNormal = LimitSwitchNormal_NormallyClosed;
.sum0Term = Integrated Sensor;
.sum1Term = Integrated Sensor;
.diff0Term = Integrated Sensor;
.diff1Term = Integrated Sensor;
.openloopRamp = 0.000000;
.closedloopRamp = 0.000000;
.peakOutputForward = 1.000000;
.peakOutputReverse = -1.000000;
.nominalOutputForward = 0.000000;
.nominalOutputReverse = 0.000000;
.neutralDeadband = 0.001000;
.voltageCompSaturation = 11.000000;
.voltageMeasurementFilter = 32;
.velocityMeasurementPeriod = VelocityMeasPeriod::Period_100Ms;
.velocityMeasurementWindow = 64;
.forwardSoftLimitThreshold = 0;
.reverseSoftLimitThreshold = 0;
.forwardSoftLimitEnable = 0;
.reverseSoftLimitEnable = 0;
.slot0.kP = 1.400000;
.slot0.kI = 0.010000;
.slot0.kD = 0.000000;
.slot0.kF = 0.000000;
.slot0.integralZone = 100;
.slot0.allowableClosedloopError = 0;
.slot0.maxIntegralAccumulator = 0.000000;
.slot0.closedLoopPeakOutput = 1.000000;
.slot0.closedLoopPeriod = 1;
.slot1.kP = 0.000000;
.slot1.kI = 0.000000;
.slot1.kD = 0.000000;
.slot1.kF = 0.000000;
.slot1.integralZone = 0;
.slot1.allowableClosedloopError = 0;
.slot1.maxIntegralAccumulator = 0.000000;
.slot1.closedLoopPeakOutput = 1.000000;
.slot1.closedLoopPeriod = 1;
.slot2.kP = 0.000000;
.slot2.kI = 0.000000;
.slot2.kD = 0.000000;
.slot2.kF = 0.000000;
.slot2.integralZone = 0;
.slot2.allowableClosedloopError = 0;
.slot2.maxIntegralAccumulator = 0.000000;
.slot2.closedLoopPeakOutput = 1.000000;
.slot2.closedLoopPeriod = 1;
.slot3.kP = 0.000000;
.slot3.kI = 0.000000;
.slot3.kD = 0.000000;
.slot3.kF = 0.000000;
.slot3.integralZone = 0;
.slot3.allowableClosedloopError = 0;
.slot3.maxIntegralAccumulator = 0.000000;
.slot3.closedLoopPeakOutput = 1.000000;
.slot3.closedLoopPeriod = 1;
.auxPIDPolarity = 0;
.remoteFilter0.remoteSensorDeviceID = 11;
.remoteFilter0.remoteSensorSource = RemoteSensorSource::RemoteSensorSource_CANCoder;
.remoteFilter1.remoteSensorDeviceID = 0;
.remoteFilter1.remoteSensorSource = RemoteSensorSource::RemoteSensorSource_Off;
.motionCruiseVelocity = 0;
.motionAcceleration = 0;
.motionCurveStrength = 0;
.motionProfileTrajectoryPeriod = 0;
.feedbackNotContinuous = 0;
.remoteSensorClosedLoopDisableNeutralOnLOS = 0;
.clearPositionOnLimitF = 0;
.clearPositionOnLimitR = 0;
.clearPositionOnQuadIdx = 0;
.limitSwitchDisableNeutralOnLOS = 0;
.softLimitDisableNeutralOnLOS = 0;
.pulseWidthPeriod_EdgesPerRot = 1;
.pulseWidthPeriod_FilterWindowSz = 1;
.trajectoryInterpolationEnable = 1;
.customParam0 = 0;
.customParam1 = 0;
```

Queried configuration afterward:
```
.supplyCurrLimit = Current Limiting will activate if SUPPLY current exceeds 30 amps for 0.1 seconds.  Then current will hold at 30 amps;
.statorCurrLimit = Limiting is disabled.;
.motorCommutation = 0;
.absoluteSensorRange = 0;
.integratedSensorOffsetDegrees = 0.000000;
.initializationStrategy = 0;
.primaryPID.selectedFeedbackSensor = RemoteSensor0;
.primaryPID.selectedFeedbackCoefficient = 1.000000;
.auxiliaryPID.selectedFeedbackSensor = Integrated Sensor;
.auxiliaryPID.selectedFeedbackCoefficient = 1.000000;
.forwardLimitSwitchSource = LimitSwitchSource_RemoteTalonSRX;
.reverseLimitSwitchSource = LimitSwitchSource_RemoteTalonSRX;
.forwardLimitSwitchDeviceID = 4;
.reverseLimitSwitchDeviceID = 0;
.forwardLimitSwitchNormal = LimitSwitchNormal_NormallyClosed;
.reverseLimitSwitchNormal = LimitSwitchNormal_NormallyClosed;
.sum0Term = Integrated Sensor;
.sum1Term = Integrated Sensor;
.diff0Term = Integrated Sensor;
.diff1Term = Integrated Sensor;
.openloopRamp = 0.000000;
.closedloopRamp = 0.000000;
.peakOutputForward = 1.000000;
.peakOutputReverse = -1.000000;
.nominalOutputForward = 0.000000;
.nominalOutputReverse = 0.000000;
.neutralDeadband = 0.000978;
.voltageCompSaturation = 11.000000;
.voltageMeasurementFilter = 32;
.velocityMeasurementPeriod = VelocityMeasPeriod::Period_100Ms;
.velocityMeasurementWindow = 64;
.forwardSoftLimitThreshold = 0;
.reverseSoftLimitThreshold = 0;
.forwardSoftLimitEnable = 0;
.reverseSoftLimitEnable = 0;
.slot0.kP = 1.400000;
.slot0.kI = 0.010000;
.slot0.kD = 0.000000;
.slot0.kF = 0.000000;
.slot0.integralZone = 100;
.slot0.allowableClosedloopError = 0;
.slot0.maxIntegralAccumulator = 0.000000;
.slot0.closedLoopPeakOutput = 1.000000;
.slot0.closedLoopPeriod = 1;
.slot1.kP = 0.000000;
.slot1.kI = 0.000000;
.slot1.kD = 0.000000;
.slot1.kF = 0.000000;
.slot1.integralZone = 0;
.slot1.allowableClosedloopError = 0;
.slot1.maxIntegralAccumulator = 0.000000;
.slot1.closedLoopPeakOutput = 1.000000;
.slot1.closedLoopPeriod = 1;
.slot2.kP = 0.000000;
.slot2.kI = 0.000000;
.slot2.kD = 0.000000;
.slot2.kF = 0.000000;
.slot2.integralZone = 0;
.slot2.allowableClosedloopError = 0;
.slot2.maxIntegralAccumulator = 0.000000;
.slot2.closedLoopPeakOutput = 1.000000;
.slot2.closedLoopPeriod = 1;
.slot3.kP = 0.000000;
.slot3.kI = 0.000000;
.slot3.kD = 0.000000;
.slot3.kF = 0.000000;
.slot3.integralZone = 0;
.slot3.allowableClosedloopError = 0;
.slot3.maxIntegralAccumulator = 0.000000;
.slot3.closedLoopPeakOutput = 1.000000;
.slot3.closedLoopPeriod = 1;
.auxPIDPolarity = 0;
.remoteFilter0.remoteSensorDeviceID = 11;
.remoteFilter0.remoteSensorSource = RemoteSensorSource::RemoteSensorSource_CANCoder;
.remoteFilter1.remoteSensorDeviceID = 0;
.remoteFilter1.remoteSensorSource = RemoteSensorSource::RemoteSensorSource_Off;
.motionCruiseVelocity = 0;
.motionAcceleration = 0;
.motionCurveStrength = 0;
.motionProfileTrajectoryPeriod = 0;
.feedbackNotContinuous = 0;
.remoteSensorClosedLoopDisableNeutralOnLOS = 0;
.clearPositionOnLimitF = 0;
.clearPositionOnLimitR = 0;
.clearPositionOnQuadIdx = 0;
.limitSwitchDisableNeutralOnLOS = 0;
.softLimitDisableNeutralOnLOS = 0;
.pulseWidthPeriod_EdgesPerRot = 0;
.pulseWidthPeriod_FilterWindowSz = 0;
.trajectoryInterpolationEnable = 1;
.customParam0 = 0;
.customParam1 = 0;
```

This is TalonFX 21.0.0.0 firmware.

CoryNessCTR, 2021-12-29:
Thanks for the issue!
This is a known issue for FX firmware 21.0.0.0 and was fixed in a recent update, which will be packaged for the season 2022 release.
In the meantime, I have attached a candidate release of our 2022 FX firmware which has this problem resolved: [TalonFX-Application-22.0.0.0.zip](https://github.com/CrossTheRoadElec/Phoenix-Releases/files/7790300/TalonFX-Application-22.0.0.0.zip)

dkt01, 2021-12-29:
@CoryNessCTR Thanks!  Is the 22.0.0.0 Falcon firmware compatible with the 21.0.0.0 Phoenix libraries and the 21 CANCoder firmware or do we need a full upgrade to the 2022 release?  My project is using the Linux socket CAN libraries for Raspberry Pi.

CoryNessCTR, 2021-12-29:
22.0.0.0 firmware should work with the 5.19.4 Phoenix libraries released during the 2021 season & 21.X CANcoder firmware, correct.

closed 2022-01-03 by CoryNessCTR