CrossTheRoadElec/Phoenix-Releases #23: WPI_PigeonIMU not compatible with the Gyro Shuffleboard Widget

IShaLinI, 2022-01-07:
The WPI_PigeonIMU class implements Sendable in a way that makes it incompatible with the Gyro Widget.

Current initSendable method:
```
@Override
public void initSendable(SendableBuilder builder) {
    builder.setSmartDashboardType("Pigeon IMU");
    builder.addDoubleProperty("Heading", this::getFusedHeading, null);
}
```

The SmartDashboardType and the key "Heading" make it not able to be displayed with the Gyro Widget.

Fixed initSendable method:
```
@Override
public void initSendable(SendableBuilder builder) {
    builder.setSmartDashboardType("Gyro");
    builder.addDoubleProperty("Value", this::getFusedHeading, null);
}
```

CoryNessCTR, 2022-01-07:
Thanks for the Issue!
This seems like a simple enough update, we've got it noted in our private tracker and will try to get it into the next release.

CoryNessCTR, 2022-02-04:
Release https://github.com/CrossTheRoadElec/Phoenix-Releases/releases/tag/v5.21.1.0 has this fixed

closed 2022-02-04 by CoryNessCTR