CrossTheRoadElec/Phoenix-Releases #24: CANCoderSimCollection.setVelocity() only supports 16 bit signed values

PaulTerrasi, 2022-01-17:
I've observed this behavior running in simulation on v5.20.2 of the phoenix library.
``` java
cancoder.getSimCollection().setVelocity(32767);
System.out.println(m_rightEncoder.getVelocity()); // 28799.12109375
cancoder.getSimCollection().setVelocity(32768);
System.out.println(m_rightEncoder.getVelocity()); // -28800.0
```

It looks like the input to `setVelocity()` is a signed 16 bit integer, and overflowing when given a value that is too high.

The CANCoder is rated for 15,000 RPM, which is 102,400 ticks per 100ms if I did my math right. It would be nice if the simulation supported those speeds.

CoryNessCTR, 2022-01-17:
Thanks for the issue!
We're able to reproduce this and tracked down the root cause.
This should get fixed in the next release which should come out later this month.

CoryNessCTR, 2022-02-04:
Release https://github.com/CrossTheRoadElec/Phoenix-Releases/releases/tag/v5.21.1.0 has this fixed

closed 2022-02-04 by CoryNessCTR