CrossTheRoadElec/Phoenix-Releases #25: Install Pheonix Unattended

reediculous456, 2022-01-28:
I use an SCCM-like solution to push software to my devices. Is there a command line flag that would allow me to install Phoenix in the background without a user clicking through the GUI?

daltzctr, 2023-06-15:
Hi, we just wanted to clarify if you are interested in installing all of Phoenix headless (Tuner + Maven Artifacts) or just Tuner? With Tuner X being distributed via the Windows Store, you can install it headless with the following command-line argument:

- `winget install "Phoenix Tuner X"`

reediculous456, 2023-06-17:
I’m looking to install the whole suite. Something like running the installer from command prompt with a flag to set the config items and install without a GUI