CrossTheRoadElec/Phoenix-Releases #28: CTRE motor controller stopMotor() api break follow() configurations

wegnergit, 2022-02-22:
**Describe the bug**
CTRE's WPI motor controllers (i.e. WPI_TalonFX, WPI_VictorSPX, WPI_TalonSRX, ...) clear follow() configuration.
Their stopMotor() api breaks follow() configurations for slave motors.  
Which means "Test" mode will reset any following motors configuration to follow, so if reenter "Teleoperated" or "Autonomous" mode and after "Test" mode none of follow motors will move.

**To Reproduce**
Steps to reproduce the behavior using "Test" mode:
1. Run simulator (with joystick 0 setup and expand devices WPI_TalonFX [1] and [2]) or robot 
2. Start in Teleoperated mode
3. See motor 1 (mater) and 2 (slave) running percentOutput at 1.0
4. Switch to Test mode -> See only master motor at speed -- percentOutput on 1 at 1 and on 2 at 0.0!!!!! (ISSUE slave motor no longer following)    

Steps to reproduce the behavior using example robot code (using commands to show it is WPI's stopMotors() method causing issue):
1. Run simulator (with joystick 0 setup and expand devices WPI_TalonFX [1] and [2]) or robot 
2. Start in Teleoperated mode
3. See motor 1 (mater) and 2 (slave) running percentOutput at 1.0
4. Press A button -> See motors stop -- both motors percentOutput at 0.0
5. Press B button -> See only master motor at half speed -- percentOutput on 1 at about .478 and on 2 at 0.0!!!!! (ISSUE slave motor no longer following)
6. Press X button -> see slave motor back to following (only after reconfiguring slave motor to follow -- refollowMotors()) 

**Expected behavior**
The slave motors configuration should be cleared after stopMotor() is executed and should not have to be reconfigured to follow (i.e. slaveMotor.follow(masterMotor);)

**Screenshots for second scenario**
After Teleoperated mode:
[image]
After Press A button (calls stopMotors() which causes issue):
[image]
After Press B button (starts master at half speed but see slave motor [2] not running):
[image]
After Press X button (reconfigures follow motor fixes following but should not have to do this):
[image]

 
**Desktop (please complete the following information):**
 - WPILib Version: 2022.3.1
 - OS: Windows 10
 - Java version 1.11 (not sure)
 - C++ version N/A (may exist with C++ too)

**Additional context**
Attaching zip file with example FRC command base robot program (WHYNOTFOLLOW) to reproduce issue as described above (see RobotContainer.java and ExampleSubsystem.java)
[src.zip](https://github.com/wpilibsuite/allwpilib/files/8119274/src.zip)
.

JCaporuscio, 2022-02-22:
This is working as intended - it is not a bug.

Follower mode is not a persistent configuration, it is a control mode.
So if you command the motor to do anything besides follow (like stop), you then need to tell it to follow again.

I recommend either calling follow() in your init functions (or anywhere periodically works, too) or only telling the master motor controllers to stop.

closed 2022-02-22 by JCaporuscio

wegnergit, 2022-02-22:
So you even think re-need to reconfigure follow during testPeriodic() given stopMotor() is continually called during Test mode.

JCaporuscio, 2022-02-22:
setSafeState() (which is stopMotor() for WPI_TalonFX) should only be called when entering and exiting Test Mode.
So calling .follow() in testInit() or testPeriodic() would be fine, this lets you have the follower setup while in Test mode.

Then make sure to call .follow() in your other init() functions.

wegnergit, 2022-02-22:
Or I would suggest reconfiguring in testExit() versus init() functions, so other modes would not be affected.

wegnergit, 2022-02-23:
SIDE NOTE: The issue is not only with the stopMotor(), but also CTRE WPI motor implementation of initSendable() and use of this::set (but I believe needed for updating follow motors values).  
```
	@Override
	public void initSendable(SendableBuilder builder) {
		builder.setSmartDashboardType("Motor Controller");
		builder.setActuator(true);
		builder.setSafeState(this::stopMotor);
		builder.addDoubleProperty("Value", this::get, this::set);
	}
```
But when set() is executed on the follow motor controller it also overwrites the mode of the follow motor to PercentOutput and nor longer follows.

So the solution is still the same:  for any following motor you will need to be re-configure to follow after "Test" mode.  And no one should set() or call stopMotor() on following motor unless they wish break the follow mode of the motor controller.