CrossTheRoadElec/Phoenix-Releases #29: Limit Switch trigger on Master should affect follower

ncgears, 2022-02-25:
The limit trigger of a master TalonSRX should also control the following device.  Currently, the limit appears to only halt the "output" side of the Master Talon, so the logic is still effectively in process and the follower Talon does not halt.

This default behavior appears (obstensibly) that it could be overridden using a remote sensor on the follower.  Doing this should be the default behavior as there is few instances where you would EVER want the follower to continue doing a thing after the master has limited and stopped its output.

JCaporuscio, 2022-02-25:
This is not the intended behavior - a follower motor should stop when the master is limited.

Can you send us an export of the motor controller configs and your code?  That would be helpful in trying to reproduce the issue.

ncgears, 2022-02-25:
Code for master/follower is at https://github.com/ncgears/NCYoda22/blob/dev/src/main/java/frc/team1918/robot/subsystems/ClimberSubsystem.java

Lines 36-51. When running the master in the forward direction, the forward limit was shorted and the follower tried to continue running (or there was a delay of enough to be concerning)

JCaporuscio, 2022-02-25:
This is the issue:
https://github.com/ncgears/NCYoda22/blob/dev/src/main/java/frc/team1918/robot/subsystems/ClimberSubsystem.java#L99

You're calling set() on the follower motor - this changes the output to PercentOutput instead of follower mode so it's not actually following.

ncgears, 2022-02-25:
> This is the issue: https://github.com/ncgears/NCYoda22/blob/dev/src/main/java/frc/team1918/robot/subsystems/ClimberSubsystem.java#L99
> 
> You're calling set() on the follower motor - this changes the output to PercentOutput instead of follower mode so it's not actually following.

I see this, retroactively, I was testing last night and reverted this change a student made, but maybe it didn't get saved?  I had changed it back to a follower and will re-test tonight and update this issue.

ncgears, 2022-02-26:
Identified cause of issue. (User error). Limit switch was connected to follower, not to master. Everything working as intended.

closed 2022-02-27 by ncgears