CrossTheRoadElec/Phoenix-Releases #30: CANdle cancels first API call if not sent?

ddaymn, 2022-03-01:
I believe this is a bug in the candle api

We are using the candle to provide green lighting for vision and then the led outputs for driver signaling.

In the same method we make these 2 calls

     candle.setLEDs(r, g, b, 0, 8, 66);
     candle.setLEDs(0, cameraLedOn ? 255 : 0, 0, 0, 0, 8);

when we do this the colors in the first call are never set.  If we reverse the lines then the 8 lights on the Candle never get lit

latest phoenix lib and firmware applied using rio canbus

as a workaround if we put a Timer.delay of .1 sec between the calls it works.

JCaporuscio, 2022-03-01:
You'll need this development release:
https://github.com/CrossTheRoadElec/Phoenix-Examples-Languages/blob/CANdle-MultiAnimations/Java%20General/CANdle%20MultiAnimation/vendordeps/Phoenix.json

And the latest CANdle firmware (22.1.1.0) here:
https://github.com/CrossTheRoadElec/Phoenix-Releases/tree/master/ctr-device-firmware

The dev release is just a small feature-improvement release based on the latest public full release - while it hasn't gone through the entire rigorous test process for a full release the specific changes are minor and well-tested so it's safe to use on your robot.  We wanted folks to be able to use the multi-animation CANdle features (required for your code snippet to work) without waiting for the next full release.

CoryNessCTR, 2022-11-30:
This has been merged into the main releases, so closing

closed 2022-11-30 by CoryNessCTR