CrossTheRoadElec/Phoenix-Releases #32: WPI Simulation not working in 5.21.3

allengregoryiv, 2022-09-22:
I was having issues getting a new WPILib project to simulate TalonFXs and couldn't figure out what was different from our old in-season project that was working correctly.

I finally looked and saw the old project was running Phoneix 5.21.1 and was simulating values correctly, but when I updated the vendordep to 5.21.3, that project also stopped working.

They now both give errors like this while running and don't update any of the values.

> "CTR: CAN frame not received/too-stale.
>         Talon SRX 21 GetMotorOutputPercent
> CTR: Firm Vers could not be retrieved. Use Phoenix Tuner to check ID and firmware(CRF) version. 
>         Talon FX 20 (3847)"

Did something change in 5.21.2 or 5.21.3 to break simulation or are there new examples to make it work correctly?

JCaporuscio, 2022-09-22:
Yep, there was an issue in 5.21.3 that broke simulated objects instantiating correctly.

There's an updated version (5.21.4) that's available if you do on online install or update of the vendordep through vscode that will fix the issue.

JCaporuscio, 2022-11-30:
Closing as this is fixed in current releases

closed 2022-11-30 by JCaporuscio