CrossTheRoadElec/Phoenix-Releases #36: Desktop build for mac fails

rwlee, 2023-01-13:
With the CTRE libraries manually installed into wpilib, using the `Phoenix5.json` and `PhoenixProAnd5.json` vendordeps, the following error occurs:
```
* What went wrong:
Could not determine the dependencies of task ':compileFrcUserProgramOsxuniversalDebugExecutableFrcUserProgramCpp'.
> Could not resolve all files for configuration ':frcUserProgramosxuniversalDebugExecutable_uberheaders'.
   > Could not find simProCANcoder-23.0.1-headers.zip (com.ctre.phoenixpro.sim:simProCANcoder:23.0.1).
     Searched in the following locations:
         file:/Users/sgreene/wpilib/2023/maven/com/ctre/phoenixpro/sim/simProCANcoder/23.0.1/simProCANcoder-23.0.1-headers.zip
   > Could not find simProPigeon2-23.0.1-headers.zip (com.ctre.phoenixpro.sim:simProPigeon2:23.0.1).
     Searched in the following locations:
         file:/Users/sgreene/wpilib/2023/maven/com/ctre/phoenixpro/sim/simProPigeon2/23.0.1/simProPigeon2-23.0.1-headers.zip
   > Could not find simProTalonFX-23.0.1-headers.zip (com.ctre.phoenixpro.sim:simProTalonFX:23.0.1).
     Searched in the following locations:
         file:/Users/sgreene/wpilib/2023/maven/com/ctre/phoenix/Users/ryanlee/frc2023/error.txtpro/sim/simProTalonFX/23.0.1/simProTalonFX-23.0.1-headers.zip
```

It's a bit weird that normal phoenix is looking for Pro resources.

If the CTRE libraries are not manually installed, please see the attached error file:
[error.txt](https://github.com/CrossTheRoadElec/Phoenix-Releases/files/10414631/error.txt)

JCaporuscio, 2023-01-13:
Couple of questions:
1. Are you using both `Phoenix5.json` and `PhoenixProAnd5.json` in your project at the same time?  Or are you saying you've tested with both of them one at a time?  Only one of those jsons should exist within a project.
2. It looks like it's only searching the local filepath for artifacts.  Are you using the --offline flag?  If not, something odd is going on with the dependency searching in your project.

> It's a bit weird that normal phoenix is looking for Pro resources.
Modern Phoenix 5 utilizes Pro's tools and simulation binaries, this way it can take advantage of low-level updates/improvements and is easier to maintain.

JCaporuscio, 2023-01-13:
Ok, so digging further I think there's two things happening here:

First, it looks like the zip files are missing the header artifacts for simulation libraries (this is a bug on our end).  So when you copied the files locally, it find the binary there and then can't find the headers, causing the error you're seeing.
The workaround is to delete the phoenix/phoenixpro folders from your local install so the build has to find the artifacts in the online repo.

Second, the error file you attached has some errors in wpimath/wpiutil that look related to C++ units.  This *may* be caused by a double-include issue that both us and photonvision ran into.  Try moving all phoenix-related includes below all the other includes in your project - this may work around the issue.

Both of these will be fixed in our next release (likely today, actually)

rwlee, 2023-01-13:
@JCaporuscio -- thanks, was able to get a build working

There are a variety of issues, but your workaround worked for the first issue.
The second one seems related to namespaces, specifically `using namespace units` and `using namespace units::time`. If there's a known issue in a public repo I would love a link so that I can dig into the context a bit. Was able to get a build working without reorganizing the includes by removing those namespace usages. There were still a few CTRE related warnings when building:
```
In file included from /Users/ryanlee/frc2023v2/src/main/cpp/subsystems/Drivetrain.cpp:1:
In file included from /Users/ryanlee/frc2023v2/src/main/include/subsystems/Drivetrain.h:25:
In file included from /Users/ryanlee/frc2023v2/src/main/include/utils/SwerveModule.h:3:
In file included from /Users/ryanlee/.gradle/caches/transforms-3/0b3d574b8cd0199fc938043568013995/transformed/api-cpp-sim-5.30.2-headers/ctre/Phoenix.h:42:
In file included from /Users/ryanlee/.gradle/caches/transforms-3/e3e9b54237a3337462df6a321ecead55/transformed/wpiapi-cpp-sim-5.30.2-headers/ctre/phoenix/motorcontrol/can/WPI_TalonFX.h:14:
/Users/ryanlee/.gradle/caches/transforms-3/e3e9b54237a3337462df6a321ecead55/transformed/wpiapi-cpp-sim-5.30.2-headers/ctre/phoenix/motorcontrol/can/WPI_BaseMotorController.h:83:17: warning: 'Get' overrides a member function but is not marked 'override' [-Winconsistent-missing-override]
        virtual double Get() const;
```

I would love to get rid of those too, but that's not the end of the world because the build was still successful.

JCaporuscio, 2023-01-13:
We'll dig into the warnings and see what we can do to fix them, not sure on an ETA though.

We did just release our first post-kickoff update - it fixes the local install header issue and may resolve the `using` issue you described as it fixes the units double-include issue I mentioned above.
https://github.com/CrossTheRoadElec/Phoenix-Releases/releases/tag/v5.30.3.0

I'm going to close this issue for now, feel free to re-open or make a new issue if you run into anything else or reproduce the `using` issue.

closed 2023-01-13 by JCaporuscio

rwlee, 2023-01-13:
Sounds good, `using` issue persists but we'll work around it for now. Thanks @JCaporuscio