CrossTheRoadElec/Phoenix-Releases #37: Build Failing Due to Reference to a Nonexistent File

ankithm28, 2023-01-15:
We're referencing the `Talon FX` motor controller. This raises an error on compilation that there is a reference to `Platform.hpp`. 

`#include "ctre/phoenix/platform/Platform.hpp"`
[image]
[image]

The documentation at `https://api.ctr-electronics.com/phoenix/release/cpp/_platform___c_c_i_8h_source.html` also references this file but doesn't exist.

Thanks for the help!

ozrien, 2023-01-15:
Header file appears to be there on my side.  I pulled [Motion Magic Example](https://github.com/CrossTheRoadElec/Phoenix-Examples-Languages/tree/master/C%2B%2B%20General/MotionMagic) to quick test and it builds (sim and rio).

My guess is the [Phoenix.json](https://github.com/CrossTheRoadElec/Phoenix-Examples-Languages/blob/master/C%2B%2B%20General/MotionMagic/vendordeps/Phoenix.json) is wrong in your project.

ozrien, 2023-01-15:
If its easy you can send us your project and we can take a quick look.

closed 2023-01-17 by ankithm28