CrossTheRoadElec/Phoenix-Releases #38: Talon commanded voltage has delayed reporting on 2023 firmware

Oblarg, 2023-02-03:
Per discussion in the FRC discord, we're seeing negative calculated values for measurement delay when users run SysId routines on 2023 Talons:

[image]

From the [SysId source](https://github.com/wpilibsuite/sysid/blob/main/sysid-application/src/main/native/cpp/analysis/FilteringUtils.cpp#L114), we can see these values are computed as follows:

[image]

Both values are computed from the step voltage test data.

* The position delay is computed as the time difference between the first application of voltage to the motor, and the first recorded motion of the motor.
* The velocity delay is computed as the time difference between the first application of voltage to the motor, and the largest acceleration seen during the test.

The simplest explanation for this is that somehow the voltage data is delayed compared to the position data.  SysId's robot binary queries the motor controller API to determine applied voltage, and this is only being observed with CTRE devices, so it seems like it might be a result of 2023 firmware changes.

sciencewhiz, 2023-02-04:
I can see CTRE needing several more pieces of information in order to investigate

What actual motor controller. Presumably you don't mean the [PWM only Talon](https://www.chiefdelphi.com/t/new-talon-speed-controller/123059) released in 2012
What version of Phoenix vendordep (current SysID releases use beta vendordeps, but there's other versions in the repo)
What are the actual calls SysID uses for retrieving the voltage and motion, as there are sometimes different methods with different update rates.

daltzctr, 2024-10-17:
We have made many enhancements to improve data collection (HootLog + our SysID swerve requests) and data reporting (enhancements on the TalonFX to improve reported data) . In our testing, it's shown improved results in SysID reported gains on the data.

https://v6.docs.ctr-electronics.com/en/latest/docs/yearly-changes/yearly-changelog.html#improved-swerve-requests

Please open a new issue if you notice any other issues or potential improvements that can be made.