CrossTheRoadElec/Phoenix-Releases #40: Phoenix pro motion magic not handling inversion properly

mjansen4857, 2023-02-05:
In simulation at least, I can't speak for a real robot yet. When using motion magic (possibly other position/velocity control modes as well), inversion is not being handled properly. 

When inverted (clockwise positive), if I set a positive target position, the motor will be driven in the correct direction (clockwise), but it will target a position in the opposite direction (negative/counter clockwise). Because the motor output and sensor value is being inverted, motion magic will be targeting a negative position while driving in the positive direction, so the target is never reached.

ozrien, 2023-02-05:
Phoenix 5 or Phoenix Pro?
Note there was  Phoenix Pro fix last week ([Change log](https://api.ctr-electronics.com/changelog) 2023/01/30) that addresses an issue with invert.

mjansen4857, 2023-02-05:
Phoenix pro. We are using the latest update. It seems to work fine on a real robot.

JCaporuscio, 2023-02-20:
Closing as complete - this was fixed in 23.0.7
https://api.ctr-electronics.com/changelog#20230210

closed 2023-02-20 by JCaporuscio