CrossTheRoadElec/Phoenix-Releases #41: Phoenix pro soft limits rapidly switching between neutral and driving past limit

mjansen4857, 2023-02-05:
In phoenix pro, after configuring the software limit switch functionality, driving the motor up to or passt the soft limit will produce some really weird behavior. The motor will reach the limit and then start rapidly switching between trying to drive further past the limit and being forced to neutral. The motor will get stuck like this for a few seconds before allowing the motor to be driven to a normal position again.

We are using the latest firmware and the latest version of phoenix pro. I've noticed this happen both in simulation and on physical hardware.

Here's a video of this happening. You can mostly tell from the audio, but the status lights go really dim, I'm assuming from switching so rapidly:

https://user-images.githubusercontent.com/9343077/216834256-443cc928-a38d-4f21-bb50-ea9d5fba1bde.mov

ozrien, 2023-02-05:
I suspect what's happening is the the rotor goes past the limit, output is
neutral, rotor relaxes, position goes back under the limit, motor output is
allowed, repeat.

Phoenix 5 and pro should both behave this way at since their is no hysteris
in either firmware.

I was thinking about adding a hysteris threshold config years ago, but
after all these years no one actually reported this.

Or maybe add a torque-current so talon switches to torque mode when
limited, and applies a user specified torque to keep it limited, which is
easy to do with pro.

Can you confirm this is the root cause by plotting the position ( use
whatever plotter you want, or just print the sensor position and watch it
over time).

mjansen4857, 2023-02-05:
It actually keeps going further past the limit while doing this. If you watch the gear in the top right of the video you can see it keeps driving past the limit. I plotted this before in simulation, and noticed that the commanded voltage would quickly switch between 0 and some output voltage (in this case a negative voltage since it is driving past the reverse limit).

I will plot the position/output later this afternoon and share that here.

ozrien, 2023-02-05:
Keep in mind there is no guarantee the mechanism stops at the limit.  It
just neutrals the motor output.  Momentum will likely carry the mechanism a
bit before it relaxes (settles).

Enabling brake might reduce that if it's not enabled already.  You can set
the config or just override in the control object.

Looking at the video I *think* brake setting is off.

mjansen4857, 2023-02-05:
It wasn't breaking, but the configuration of the mechanism in the video was capable of holding its position with neutral output/power off. 

I was mainly talking about it going past the limit in response to this:
> I suspect what's happening is the the rotor goes past the limit, output is
neutral, rotor relaxes, position goes back under the limit, motor output is
allowed, repeat.

So, its not continuously going past the limit and back within it. Some more context on what was being tested here that I forgot to mention: We had a reverse limit set to some value (I think about -25 degrees), then, we gave it a motion magic control to something past the limit, like -30 degrees. Once it reaches the soft limit, it will start freaking out like in the video, and keep trying to drive toward the target. If we give it a new motion magic command that is within the limits as soon as it starts (0 degrees), it will stay stuck in the oscillation for a few seconds before finally driving to that new position. It seems like what may be happening with the delay is that MM sees the negative velocity caused by the oscillation, and keeps driving it against the limit as it tries to accelerate toward a positive velocity, causing the weird delay since it can't keep driving in the negative direction. I'm not sure if MM works this way, but it sounds like it would make sense.

mjansen4857, 2023-02-05:
Here's a plot of what's happening:
[image]

Yellow line/left axis is the rotor position (scaled to degrees), blue line/right axis is voltage output to the motor. The graph starts by driving the motor in the forward direction towards a target within the limits, then, drives in reverse towards a target outside of the limit. As you can see in the voltage plot, it is oscillating between neutral and negative voltage, while the rotor position continues to go slightly further past the limit. The limit in this case is -20.

mjansen4857, 2023-02-20:
Is there any fix in the works for this? We've had to disable the soft limit feature and implement it ourselves for the robot to be usable.

JCaporuscio, 2023-02-20:
We've been able to to reproduce the issue and believe we've isolated the root cause.
We'll try to provide another update in the next day or two - I don't want to provide a timing estimate on a fix until we've done more testing.

CoryNessCTR, 2023-02-22:
Fix is available in firmware 23.6.6.1
Let us know if you have any other questions

https://api.ctr-electronics.com/changelog#20230222

closed 2023-02-22 by CoryNessCTR