CrossTheRoadElec/Phoenix-Releases #43: Phoenix Pro Orchestra implementation

Benjamin-GRS, 2023-02-22:
The current TalonFXs in PhoenixPro are not compatible with the Orchestra object in the v5 version. We would like to be able to use this feature with the current firmware and have the other benefits of the API.

JCaporuscio, 2023-02-27:
This is something that will likely be implemented in some form in the future, but it will not be available during this season.

InterstellarDrift, 2023-09-24:
It would appear that the Orchestra has now been ported to phoenix 6 and is available in the 2024 open alpha. It is referenced in the release notes and is available in the code!

JCaporuscio, 2023-09-25:
It is!

For anyone else curious, see our alpha announcement here:
https://store.ctr-electronics.com/blog/2024-phoenix-6-alpha/

If you have a chance to use the alpha, please let us know your thoughts and any issues you encounter.

This issue tracker will remain open until the next public stable release that includes Orchestra, so feel free to leave thoughts here or at the alpha email provided in the announcement.

closed 2023-12-06 by ozrien