CrossTheRoadElec/Phoenix-Releases #47: Using VelocityVoltage() on a TalonFX in unit tests does not update TalonFXSimState.getMotorVoltage()

jonathandao0, 2023-09-27:
**Setup**
Using the 2024 Phoenix 6 Alpha Library in Java, with the Unit Testing example found here: https://pro.docs.ctr-electronics.com/en/latest/docs/api-reference/wpilib-integration/unit-testing.html

**Issue**
When I modified the unit-testing example to set the TalonFX to use `VelocityVoltage()` and then check `TalonFXSimState.getMotorVoltage()`, `TalonFXSimState.getMotorVoltage()` returns 0. Reverting it back to use `DutyCycleOut()`, `TalonFXSimState.getMotorVoltage()` returns a value. This issue appears to only affect unit tests, as using `VelocityVoltage()` in simulation appears to update `TalonFXSimState.getMotorVoltage()` properly.

CoryNessCTR, 2023-09-27:
Did you set the PID gains at the start of the unit test? If I remember
correct, the devices are constructed cleanly in our example unit tests, so
you'll have to set PID gains in your unit test to get a response when using
closed loop.

jonathandao0, 2023-09-27:
Oh, guess that was it. Thanks!

closed 2023-09-27 by jonathandao0