CrossTheRoadElec/Phoenix-Releases #52: Add Python Simulation Support to Phoenix 6

lospugs, 2023-11-14:
I was attempting to test out the Phoenix 6 Python API by porting the java CommandBasedDrive example to Python to try out the beta. 
Missing method: set_position from TalonFXConfigurator
Missing Classes: TalonFXSimState

I then ported some older code that simulates well (in that the robot moves on the screen, etc) with the robotpy Phoenix5 just to try out the beta. 

Working and not working code is here for reference:
https://github.com/Lambda-Corps/2024_Python_beta

JCaporuscio, 2023-11-15:
Yep - the simulation functions in Python are on our near-term roadmap.
No exact ETA I can provide yet but it should be coming soon.

bhall-ctre, 2023-12-06:
This has been added with the release of Phoenix 6 2024 beta 3: https://api.ctr-electronics.com/changelog#20231122

closed 2023-12-06 by ozrien