CrossTheRoadElec/Phoenix-Releases #53: Tuner X Swerve Generator - Generate Offline Project

jonathandao0, 2023-11-17:
Is there a particular reason why the Tuner X Swerve Generator requires access to a physical robot to generate a project? Can it be updated to work offline where you can just enter the CAN IDs and default the encoder offsets to 0? 

Otherwise, is there an example of the `CommandSwerveDrivetrain` code generated by the Swerve Generator available somewhere as documentation? I'm trying to update my team's Phoenix 5 swerve code to Phoenix 6 and want to compare it with the generated swerve code to see if there are changes we would like to integrate into our own code.

CoryNessCTR, 2023-11-17:
The generator is more than just a configurator, it also tests that the configuration is correct, hence the need for some feedback from the device to ensure the generated code will behave properly.

However, you can run your robot code in simulation and use the Swerve Generator with the simulated robot devices.

We also have an example available here:
https://github.com/CrossTheRoadElec/Phoenix6-Examples/tree/main/java/SwerveWithPathPlanner

If you want to run in simulation, I'd even suggest using that swerve example in simulation as it contains all the necessary devices for Swerve Generation and the simulation support in the example should allow the devices to behave similarly to a real robot.

daltzctr, 2024-08-06:
Closing this issue as resolved. If you have further comments, feel free to place them here or open a new issue.

closed 2024-08-06 by daltzctr