CrossTheRoadElec/Phoenix-Releases #55: units/ headers in the api-cpp headers binary are out of sync with allwpilib

jkuszmaul, 2023-11-23:
There appears to be a `units/` folder with a bunch of C++ headers in https://maven.ctr-electronics.com/release/com/ctre/phoenix6/api-cpp/24.0.0-beta-3/api-cpp-24.0.0-beta-3-headers.zip in the latest beta release. This includes a `#include` that attempts to depend on `frc/fmt/Units.h`, which exists neither in that zip file nor in upstream allwpilib (it was moved in https://github.com/wpilibsuite/allwpilib/pull/5358/files#diff-e013f85188d8ea53a739540987ba0fdaa96e4dffaac6bff543f5eefe84ac5e84). I'm not sure if anything actually is _using_ that `units/` folder any more, or if we are just running into oddities due to not using gradle.

bhall-ctre, 2023-11-28:
The `api-cpp-*.zip` file is meant for non-FRC users who don't have access to WPILib. FRC users should use https://maven.ctr-electronics.com/release/com/ctre/phoenix6/wpiapi-cpp/24.0.0-beta-3/wpiapi-cpp-24.0.0-beta-3-headers.zip instead, which does use the WPILib units library.

With that said, "frc/fmt/Units.h" does not exist outside of FRC, so we can get that fixed in the next beta release for non-FRC users.

jkuszmaul, 2023-11-28:
Ah, thank you. I'll fix our dependency tree then. Not sure why we started using the `api-cpp` ones in the first place.

bhall-ctre, 2023-12-06:
This has been fixed in our 2024 beta 4 release: https://api.ctr-electronics.com/changelog#20231205

closed 2023-12-06 by ozrien