CrossTheRoadElec/Phoenix-Releases #56: Specify units in Java docs for SwerveRequest classes

gcschmit, 2023-11-27:
For the SwerveRequest inner classes that refer to velocities, the Java docs for the fields and methods don't specify the expected units, which I believe are m/s. The rotational rate does specify units of radians/s. These few classes stood out as units are so clearly documented elsewhere.

bhall-ctre, 2023-11-28:
> the Java docs for the fields and methods don't specify the expected units, which I believe are m/s

That is correct, the velocities are in m/s. The documentation will be updated to mention this in the next beta release.

bhall-ctre, 2023-12-06:
This has been fixed in the 2024 beta-4 release of Phoenix 6: https://api.ctr-electronics.com/phoenix6/release/java/com/ctre/phoenix6/mechanisms/swerve/SwerveRequest.FieldCentric.html#VelocityX

closed 2023-12-06 by ozrien