CrossTheRoadElec/Phoenix-Releases #61: Python phoenix6 package is not type-checkable

auscompgeek, 2024-01-06:
We're unable to run the mypy type checker on our robot code after migrating to `phoenix6`. Example output:

```
Error: components/chassis.py:3:1: error: Skipping analyzing "phoenix6": module is installed, but missing library stubs or py.typed marker  [import-untyped]
components/chassis.py:3:1: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
```

Could we please get a `py.typed` marker file in the package, per [PEP 561](https://peps.python.org/pep-0561/)?

CoryNessCTR, 2024-01-09:
This is fixed in the 24.1.0 release

auscompgeek, 2024-01-13:
I'm not seeing a `py.typed` in the `phoenix6` package in 24.1.0.

auscompgeek, 2024-02-10:
I can confirm phoenix6 24.2.0 has a `py.typed`; thanks!

closed 2024-02-10 by auscompgeek