CrossTheRoadElec/Phoenix-Releases #69: Pigeon2 phoenix hardware doesn't seem to have all the required functions.

Lagor845, 2024-01-24:
We were hoping to try python this year for our swerve implementation but I can't seem to find any documentation for how to grab rotation values for the x axis of the pigeon2. For java you bring in the pigeon class from hardware and then run the .getRotation2d to grab its rotation. It seems that the python implementation has no such function.

I cannot find the .reset or the .getRotation2d
Are the classes just developed differently? Or does the python just not have that functionality yet?

This is the Java docs
https://api.ctr-electronics.com/phoenix6/release/java/com/ctre/phoenix6/hardware/Pigeon2.html

This is the Python docs
https://api.ctr-electronics.com/phoenix6/release/python/autoapi/phoenix6/hardware/core/core_pigeon2/index.html#phoenix6.hardware.core.core_pigeon2.CorePigeon2

Thank you for all of your help!

CoryNessCTR, 2024-01-24:
Both `getRotation2d()` and `reset()` are WPILib specific classes, which we haven't yet brought over to Python.
Our Python API does however have the necessary functions to replicate this functionality, with [`get_yaw()`](https://api.ctr-electronics.com/phoenix6/release/python/autoapi/phoenix6/hardware/core/core_pigeon2/index.html#phoenix6.hardware.core.core_pigeon2.CorePigeon2.get_yaw) and [`set_yaw()`](https://api.ctr-electronics.com/phoenix6/release/python/autoapi/phoenix6/hardware/core/core_pigeon2/index.html#phoenix6.hardware.core.core_pigeon2.CorePigeon2.set_yaw)

Equivalent methods would be the following:
```python
def get_rotation2d(pigeon: Pigeon2):
    return Rotation2d.fromDegrees(pigeon.get_yaw().value)

def reset(pigeon: Pigeon2):
    pigeon.set_yaw(0)
```

> grab rotation values for the x axis

Are you sure you mean the X axis? That corresponds to the Roll Axis in Pigeon 2, which is not what `getRotation2d()` does in other languages. That method gets the Z axis, corresponding to Yaw about Pigeon 2.

Lagor845, 2024-01-24:
> Both getRotation2d() and reset() are WPILib specific methods, which we

Thank you! That one had me confused for a bit! I will inform the team and
our progress shall commence! Thank you for all you do!

daltzctr, 2024-10-16:
Fixed in beta 1 (which the python portion is pending on robotpy)

closed 2024-10-16 by daltzctr