CrossTheRoadElec/Phoenix-Releases #70: Swerve generator validation tests not running

1410n, 2024-01-25:
I've been trying to use the swerve generator in tuner x,  in the validation step the tests didn't run at all 
tuner x didn't try to run a temperry server at any point , or changing the robot code at all
and there wasn't any start test button in the GUI. 

Putting the robot in enable just run the previous robot code

bhall-ctre, 2024-02-17:
It's likely that the validation test start/stop buttons were getting clipped due to the window size being too short. We've released a Tuner X update (2024.6.2.0) that fixes the GUI layout of the validation tests so the buttons are visible on low-resolution monitors.

daltzctr, 2024-07-15:
As Ben mentioned, we did fix some issues about buttons getting clipped on small screen resolutions. If this wasn't your problem, or you notice any other issues, feel free to reopen this issue or create a new issue.

closed 2024-07-15 by daltzctr