CrossTheRoadElec/Phoenix-Releases #79: Feature Request: WPILib Measures getters and setters

TaylerUva, 2024-04-01:
WPILib provides a Units and Measures class. Phoenix classes should provide getters and setters for WPILib Measures allowing the conversions to be done in the back end.

For example TalonFX.setPosition is accepts a position in rotations as a double: 
```java
StatusCode com.ctre.phoenix6.hardware.core.CoreTalonFX.setPosition(double newValue)
```
If it instead accepts a Measure<Angle> a team could provide a value in degrees or radians the back end would automatically convert it to Rotations.

This would work great in combination with the existing `Feedback.SensorToMechanismRatio` as TalonFX.getPosition() could be returned in Unit.Rotations and be easily converted to degrees
```java
StatusSignal<Double> com.ctre.phoenix6.hardware.core.CoreTalonFX.getPosition()
```

The overall goal of this request is reduce the amount of unit conversion errors.

JCaporuscio, 2024-04-01:
This is something we'll probably look at in the off-season - `Units` and `Measures` weren't available in WPILib early enough for us to evaluate implementing it for use this season.

We do want units to be easily manageable though, so this is a good candidate for potential integration over the summer.

daltzctr, 2024-10-16:
Fixed in beta 1

closed 2024-10-16 by daltzctr