CrossTheRoadElec/Phoenix-Releases #80: Add optional drive motor torque-current feedforwards to SwerveRequest.ApplyChassisSpeeds

mjansen4857, 2024-09-26:
2025 changes to PathPlannerLib will include outputting torque-current feedforwards from path following commands and a setpoint generator utility. It appears that the Phoenix 6 Swerve mechanism's SwerveRequest types do not currently allow for passing along torque-current feedforwards with the desired ChassisSpeeds. Therefore, Phoenix Pro users utilizing the Phoenix 6 swerve template/project generator will be unable to utilize these feedforwards.

CoryNessCTR, 2024-09-27:
I think it makes more sense for PathPlanner to provide module target linear accelerations. We already have most of the parameters that are necessary to convert linear acceleration to rotor torque current (wheel radius, drive gearing, motor kT), so it reduces the burden on the user plumbing those through.

Otherwise, this sounds like a useful inclusion that we'll look into adding.

mjansen4857, 2024-09-27:
PathPlanner generation changes already require all of the values required to convert accelerations/linear forces to torque current, so there isn’t anything extra needed to support that output. I can however look into returning linear forces and accelerations as well, so the user can choose the best option for them.

daltzctr, 2024-10-17:
This has been resolved.

Please take a look at our changelog for a brief explanation and a link to API docs https://v6.docs.ctr-electronics.com/en/latest/docs/yearly-changes/yearly-changelog.html#improved-swerve-requests

closed 2024-10-17 by daltzctr