CrossTheRoadElec/Phoenix-Releases #85: TalonFX Simulation Problem

Crossle86, 2024-12-18:
The attached Java class is a wrapper for Phoeix 6 TalonFX that adds support for simulation.

This class is instantiated in my drivebase class and this classes simulationPeriodic is called
from drivebase simulationPeriodic. This classes set() function is called from my driving
command.

When running, the set() function shows the expected speed, say 1.0 but the get() function 
always returns zero. Likewise the getMotorVoltage() function always returns zero.

So I put in a constant for setInput() and then motorSimModel.getAngularPositionRotations()  does
return a rotations value. However, when I try to set that value with setRawRotorPosition(),
the StatusCode returns "simDeviceNotFound".

Additionally, when I call getPosition().getValueAsDouble() I get canbus errors in the console log saying
the device is not found. Is this expected under sim?
[Talon_FX.txt](https://github.com/user-attachments/files/18172902/Talon_FX.txt)

bhall-ctre, 2024-12-18:
We cannot reproduce the issue in a simple robot project using our latest 2025 beta. Robot class used is shown below:
```java
public class Robot extends TimedRobot {
    private final Talon_FX m_fx = new Talon_FX(50);

    Robot() {
        m_fx.getConfigurator().apply(new TalonFXConfiguration());
    }

    @Override
    public void teleopPeriodic() {
        m_fx.set(Math.sin(Utils.getCurrentTimeSeconds()));
    }

    @Override
    public void simulationPeriodic() {
        m_fx.simulationPeriodic();
    }
}
```

Is it possible your device ID is invalid? Device ID must be 0-62. Otherwise, I suspect there's some other issue with your code.

Crossle86, 2024-12-18:
I thought it would be something simple I did not know...and sure enough, I did not know about the device id range. I used 99.

So now its working but a couple questions:

I see that when I go from zero to 1.0 speed on the set() call, pretty much instantaneously, the get() function shows the current speed setting ramping up from zero to 1.0 over time. I thought get() reflects the last call to set().

Next, in your sample you set a TalonFXConfiguration but don't set any parameters of that configuration. Why is that?

bhall-ctre, 2024-12-18:
> I see that when I go from zero to 1.0 speed on the set() call, pretty much instantaneously, the get() function shows the current speed setting ramping up from zero to 1.0 over time. I thought get() reflects the last call to set().

In Phoenix 6, `get()` returns the actual duty cycle applied to the motor; it is equivalent to calling `getDutyCycle().getValue()`. Configs like current limiting and ramp rates affect the applied duty cycle, resulting in the ramping behavior you're seeing (which matches behavior with hardware).

> Next, in your sample you set a TalonFXConfiguration but don't set any parameters of that configuration. Why is that?

Device configs are persistent, even in simulation (where they are only persistent to that specific robot project). So to eliminate any issues caused by bad configs (such as setting peak output to 0), I wanted to factory default the simulated TalonFX in my test. In your robot code, you would instead set the configuration parameters on the object and call `apply()` once with that configured object.

Crossle86, 2024-12-18:
Thanks. Looking good now. I am seeing a file in my project: ctre_sim\Talon FX vers. C - 050 - 0 - ext.dat written on each sim run. Is that expected?

bhall-ctre, 2024-12-18:
Yes, that file stores persistent state of the simulated device.

daltzctr, 2024-12-19:
All of our devices persist configuration between power-off and this is an automatic behavior.

closed 2024-12-31 by Crossle86