CrossTheRoadElec/Phoenix-Releases #86: Swerve configurator has jittery steer motors caused by non-zero KV

mjansen4857, 2025-01-08:
With a KV value for the steer motor config in TunerConstants.java, the steer motors will be very jittery while steering, oscillating between forward and reverse output. This happens in both `Position` and `MotionMagicExpo` closed loop modes. This happens both with the default KV from the swerve generator, and the actual KV of the mechanism measured with sysID. Setting the steer KV to 0 resolves this issue and prevents the jitter. This leads me to believe that the KV value is being used as some sort of feedforward, even in the `Position` control mode, which shouldn't be using any feedforwads except for KS.

bhall-ctre, 2025-01-10:
> This leads me to believe that the KV value is being used as some sort of feedforward, even in the `Position` control mode

That is correct, kV is used for both `Position` and `MotionMagicExpo` steer control. In the case of `Position` control, we calculate a velocity target to supply to the position request. We added this during beta based on community feedback about the quality of steer motor tracking.

When there is no kV, the steer motors can find themselves lagging quite a bit behind their targets. This is especially true when translating while rotating in field centric control, where this manifests as noticeable translational drift. Adding a kV term allows the modules to track the rotational velocity of the robot during the continuous motion, keeping them aligned with their targets and eliminating the translational drift. We originally used `MotionMagicExpo` to accomplish this, but we decided to update `Position` to also support kV, since for both control types there are situations where one outperforms the other.

> The steer motors will be very jittery while steering, oscillating between forward and reverse output. This happens in both `Position` and `MotionMagicExpo` closed loop modes.

If it's happening for both Position and MotionMagicExpo, that sounds like kS, kV, and/or kP is too high, so you can try reducing some of those values. Keep in mind that optimal gains will result in jitter when on blocks; what matters is the behavior on the carpet.

For kS, keep in mind that when using `StaticFeedforwardSignValue.UseClosedLoopSign`, you may need to lower kS a little bit to avoid dithering. Since we calculate velocity targets for `Position`, you might be able to switch it back to `StaticFeedforwardSignValue.UseVelocitySign`, which may help the issue.

bhall-ctre, 2025-01-10:
> I'd suggest adding a comment to the withKV javadoc that explains that it will even be used in position control mode, which is currently only described as a "Velocity Feedforward Gain". I don't think anyone that doesn't know about it would assume that velocity targets are being calculated in position mode.

Just to clarify, this isn't something special done in firmware for Position control. kV is still a velocity feedforward gain, it's only applied in position control when you supply a non-zero velocity to [`withVelocity()`](https://api.ctr-electronics.com/phoenix6/release/java/com/ctre/phoenix6/controls/PositionVoltage.html#withVelocity(double)) on a `Position*` control request. We also cover this use case in the [Motion Profiling](https://v6.docs.ctr-electronics.com/en/stable/docs/api-reference/device-specific/talonfx/basic-pid-control.html#motion-profiling) section of our documentation. Just as the kA "Acceleration Feedforward Gain" can apply in position and velocity control, so too can kV.

We could update our swerve API documentation to mention that we use kV in Position control, although I'm not sure where to place that.

> Also, the default KP from the swerve generator is way too high for that behavior. It defaulted to 100, and we had to turn it down to 50 w/ no KV to avoid jittering while on the carpet.

Out of curiosity, what is your steer gear ratio, and are you using Phoenix Pro? The default kP is closer to what should be used with Phoenix Pro. It should still be somewhat reasonable without, but kP will need to be tuned down to some extent without Pro, which is why we document that users should tune their gains. SysId should be able to give you kP and kD.

mjansen4857, 2025-01-10:
We are testing on SDS mk4 modules, so the ratio is 12.8:1. We are using phoenix pro. Curiously, we used phoenix pro + position control mode last year as well, and our PID gains were nearly identical to the current swerve generator defaults (with no kv). But with 2025 code, we need a significantly lower P gain on the exact same modules. Is there any other feedforward that is being applied even with kv & ks set to 0? Not sure what else would explain this.

CoryNessCTR, 2025-01-10:
Is the robot the same between the 2024 and 2025 code? If it's significantly lighter, the wheel tread has less traction, or the floor surface is different that could explain why the gains are now underdamped.

mjansen4857, 2025-01-10:
Yeah it's the same robot, same wheels, and same carpet.

closed 2025-02-04 by mjansen4857