CrossTheRoadElec/Phoenix-Releases #90: Simulated output voltage delayed by about 100ms

mjansen4857, 2025-01-25:
When simulating a mechanism using Phoenix 6/TalonFX, the output voltage of the motor appears to be delayed by about 100ms compared to what it should be. In the following screenshot, I am running a `PositionVoltage` control mode with only kP configured. The yellow line represents the value of the closed loop reference, the blue line is the current position, and the red line is the output voltage. When the position reaches the reference, there is still a significant voltage being applied, in this case ~7V. The output voltage does not reach 0 until around 100ms later, at which point the simulation has caused the mechanism's position to overshoot from the delayed voltage readings.

[image]

For more context, we are taking the value of the `getMotorVoltage` status signal, feeding it into a linear system sim, then updating the motor's position/velocity via its sim state. There is nothing in our setup here that would cause this much of a delay.

mjansen4857, 2025-01-25:
After more testing, I believe this is actually a delay in the position set through the SimState getting to the motor's PID controller. If I zero out the PID constants and apply an arbitrary voltage feedforward to the same position request, the output voltage reacts immediately. Its only when actually calculating its output from the sim state's position that its delayed.

CoryNessCTR, 2025-01-25:
> we are taking the value of the getMotorVoltage status signal, feeding it into a linear system sim

Make sure you're using the [SimState](https://api.ctr-electronics.com/phoenix6/release/java/com/ctre/phoenix6/sim/TalonFXSimState.html) members when updating the Simulation device, the `get*()` methods will be delayed by about the same amount you'll see on a real robot, which could add up. The SimState values are nearly instantaneous in comparison.

I could see the latency involved in getting position and voltage then later updating the physics sim object could be adding up and causing what you're seeing.

mjansen4857, 2025-01-25:
Yeah that seems to have been the issue, my bad. Thanks!

closed 2025-01-25 by mjansen4857