CrossTheRoadElec/Phoenix-Releases #100: Python: phoenix6.utils.is_simulation() Always Returns True

mfisher31, 2025-02-23:
From a Tuner X generated python project, running the following code with `robotpy sim`:

```python
def robotInit(self) -> None:
    print (f"Is simulation {phoenix6.utils.is_simulation()}")
    # .. redacted 
```

... produces in the console ...
```
Is simulation False
..
```

I've been debugging why team 9431's simulator is flooding the console with CAN and firmware errors.
Our Python Code: https://github.com/snidercs/frc-bot-2025/tree/main/pybot

Full log on starup (without the print statement above).  

```
$ robotpy sim
08:35:01:416 INFO    : faulthandler        : registered SIGUSR2 for PID 183309
08:35:01:521 INFO    : halsim_gui          : WPILib HAL Simulation 2025.3.1.1
HAL Extensions: Attempting to load: libhalsim_gui
Simulator GUI Initializing.
Simulator GUI Initialized!
HAL Extensions: Successfully loaded extension
08:35:01:901 INFO    : pyfrc.physics       : Physics support successfully enabled
08:35:01:902 INFO    : wpilib              : RobotPy version 2025.3.1.1
08:35:01:902 INFO    : wpilib              : WPILib version 2025.3.1.1
08:35:01:902 INFO    : wpilib              : Running with simulated HAL.
08:35:01:904 INFO    : nt                  : Listening on NT3 port 1735, NT4 port 5810
Not loading CameraServerShared
[phoenix] CANbus Connected: can0

[phoenix] CANbus Network Down: can0

CAN frame not received/too-stale. Check the CAN bus wiring, CAN bus utilization, and power to the device.
        talon fx 2 ("") Status Signal Position
CAN frame not received/too-stale. Check the CAN bus wiring, CAN bus utilization, and power to the device.
        ctre::phoenix6::BaseStatusSignal::RefreshAll
CAN frame not received/too-stale. Check the CAN bus wiring, CAN bus utilization, and power to the device.
        talon fx 9 ("") Status Signal Position
CAN frame not received/too-stale. Check the CAN bus wiring, CAN bus utilization, and power to the device.
        talon fx 8 ("") Status Signal Position
CAN frame not received/too-stale. Check the CAN bus wiring, CAN bus utilization, and power to the device.
        talon fx 5 ("") Status Signal Position
CAN frame not received/too-stale. Check the CAN bus wiring, CAN bus utilization, and power to the device.
        pigeon 2 1 ("") Status Signal Yaw
Could not transmit CAN Frame.
        talon fx 2 ("") Apply Config
Talon ID 2 failed config with error TX_FAILED
Could not transmit CAN Frame.
        talon fx 6 ("") Apply Config
Talon ID 6 failed config with error TX_FAILED
Could not transmit CAN Frame.
        cancoder 12 ("") Apply Config
Encoder ID 12 failed config with error TX_FAILED
Could not transmit CAN Frame.
        talon fx 9 ("") Apply Config
Talon ID 9 failed config with error TX_FAILED
Could not transmit CAN Frame.
        talon fx 3 ("") Apply Config
Talon ID 3 failed config with error TX_FAILED
Could not transmit CAN Frame.
        cancoder 11 ("") Apply Config
Encoder ID 11 failed config with error TX_FAILED
Could not transmit CAN Frame.
        talon fx 8 ("") Apply Config
Talon ID 8 failed config with error TX_FAILED
Could not transmit CAN Frame.
        talon fx 7 ("") Apply Config
Talon ID 7 failed config with error TX_FAILED
Could not transmit CAN Frame.
        cancoder 10 ("") Apply Config
Encoder ID 10 failed config with error TX_FAILED
[phoenix] Library initialization is complete.

Could not transmit CAN Frame.
        talon fx 5 ("") Apply Config
Talon ID 5 failed config with error TX_FAILED
Could not transmit CAN Frame.
        talon fx 4 ("") Apply Config
Talon ID 4 failed config with error TX_FAILED
Could not transmit CAN Frame.
        cancoder 13 ("") Apply Config
[phoenix] CANbus Network Down: can0

Encoder ID 13 failed config with error TX_FAILED
Warning: Joystick Button 1 missing (max 0), check if all controllers are plugged in
  File "/home/.local/lib/python3.10/site-packages/wpilib/_impl/start.py", line 247, in _start
    self.robot.startCompetition()

  File "/home/workspace/snidercs/frc-bot-2025/pybot/robot.py", line 26, in robotInit
    self.container = RobotContainer()

  File "/home/workspace/snidercs/frc-bot-2025/pybot/robotcontainer.py", line 57, in __init__
    self.configureButtonBindings()

  File "/home/workspace/snidercs/frc-bot-2025/pybot/robotcontainer.py", line 85, in configureButtonBindings
    self._joystick.a().whileTrue(self.drivetrain.apply_request(lambda: self._brake))

  File "/home/.local/lib/python3.10/site-packages/commands2/button/trigger.py", line 158, in whileTrue
    state = SimpleNamespace(pressed_last=self._condition())

  File "/home/.local/lib/python3.10/site-packages/commands2/button/commandxboxcontroller.py", line 104, in <lambda>
    return Trigger(loop, lambda: self._hid.getAButton())

********** Robot program startup complete **********
Warning: Joystick Button 1 missing (max 0), check if all controllers are plugged in
  File "/home/.local/lib/python3.10/site-packages/wpilib/_impl/start.py", line 247, in _start
    self.robot.startCompetition()

  File "/home/.local/lib/python3.10/site-packages/commands2/commandscheduler.py", line 259, in run
    loopCache.poll()

  File "/home/.local/lib/python3.10/site-packages/commands2/button/trigger.py", line 162, in _
    pressed = self._condition()

  File "/home/.local/lib/python3.10/site-packages/commands2/button/commandxboxcontroller.py", line 104, in <lambda>
    return Trigger(loop, lambda: self._hid.getAButton())

Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        talon fx 2 ("")
Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        talon fx 6 ("")
Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        talon fx 9 ("")
Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        talon fx 3 ("")
Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        talon fx 8 ("")
Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        talon fx 7 ("")
Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        talon fx 5 ("")
Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        talon fx 4 ("")
Device firmware could not be retrieved. Check that the device is running v6 firmware, the device ID is correct, the specified CAN bus is correct, and the device is powered.
        pigeon 2 1 ("")
CAN frame not received/too-stale. Check the CAN bus wiring, CAN bus utilization, and power to the device.
        ctre::phoenix6::BaseStatusSignal::RefreshAll
[phoenix-diagnostics] Server 2025.1.0 (Feb 21 2025, 01:06:56) running on port: 1250

Warning: Joystick Button 1 missing (max 0), check if all controllers are plugged in
  File "/home/.local/lib/python3.10/site-packages/wpilib/_impl/start.py", line 247, in _start
    self.robot.startCompetition()

  File "/home/.local/lib/python3.10/site-packages/commands2/commandscheduler.py", line 259, in run
    loopCache.poll()

  File "/home/.local/lib/python3.10/site-packages/commands2/button/trigger.py", line 162, in _
    pressed = self._condition()

  File "/home/.local/lib/python3.10/site-packages/commands2/button/commandxboxcontroller.py", line 104, in <lambda>
    return Trigger(loop, lambda: self._hid.getAButton())

[phoenix] CANbus Network Down: can0

^C08:35:08:092 ERROR   : robotpy             : THIS IS NOT AN ERROR: The user hit CTRL-C to kill the robot
Traceback (most recent call last):
  File "/home/.local/lib/python3.10/site-packages/wpilib/_impl/start.py", line 127, in run
    hal.runMain()
KeyboardInterrupt

Locals at innermost frame:

{ '_start': <function RobotStarter.run.<locals>._start at 0x711eb0f6e680>,
  'retval': False,
  'robot_cls': <class 'robot.MyRobot'>,
  'rval': [False],
  'self': <wpilib._impl.start.RobotStarter object at 0x711eba1ac2e0>,
  'th': <Thread(RobotThread, started daemon 124375042885184)>}

08:35:08:092 INFO    : robotpy             : Exiting because of keyboard interrupt
08:35:08:092 INFO    : robotpy             : Robot code exited
[phoenix-diagnostics] Server shutdown cleanly. (dur:0)

[phoenix] Library shutdown cleanly

[phoenix] CANbus Disconnected: can0
```

bhall-ctre, 2025-02-23:
It looks like you're running hardware-attached sim instead of regular simulation. Make sure your phoenix6 is up to date (25.3.0.post1). You can also force simulation to run by setting the `CTR_TARGET=Simulation` environment variable.

mfisher31, 2025-02-23:
Oh nice, thanks for the tip!  That seems to have fixed the CAN warnings.  Appreciate it.

closed 2025-02-23 by mfisher31