CrossTheRoadElec/Phoenix-Releases #120: Add support for logging signal data externally

spacey-sooty, 2025-07-30:
Something similar to what URCL does where you get a Supplier<ByteBuffer[]> would be great for keeping things within the same log file.

bhall-ctre, 2025-08-04:
Unfortunately, hoot file data cannot be directly written to a WPILOG file and needs to be separately converted by Tuner X or our `owlet` CLI (which is how AdvantageScope opens hoot files). Additionally, the converted WPILOG can be more than 10x larger than the original hoot file format, which makes it impractical on the roboRIO given storage and compute limitations.

With that said, we can go in the other direction and expand support for custom user signals to include things like struct/protobuf, and we can look into better integration with WPILib APIs such as Epilogue.

jwbonner, 2025-09-13:
> we can look into better integration

From an AdvantageKit perspective, I will say that I have serious doubts about this solution. AdvantageKit’s ability to process log files relies on an open format like WPILOG, which can be adapted to our specific requirements. A closed and specialized format like Hoot does not offer the type of flexibility we need for full integration. While smaller in scale, I also expect there would be similar issues with other frameworks like Epilogue regarding metadata/timestamps/etc. AdvantageScope already needs to be designed around the strange quirks of Phoenix data generated from Hoot logs, even without needing to handle interactions with third-party frameworks.

Even if those challenges are solvable in theory, I'm also just struggling to see the advantages of moving the FRC software ecosystem further away from a standardized and adaptable format like WPILOG. Keep in mind the existing WPILOG format already supports logging arbitrary data with any kind of custom protocols/compression methods.

bhall-ctre, 2026-09-18:
This has been further improved in `26.70.0-alpha-2` ([changelog](https://api.ctr-electronics.com/changelog#20260918)) with a `HootTelemetryBackend` to integrate signal logs with the WPILib Telemetry API, and there is now a single hoot file containing multiple CAN buses (instead of a file per CAN bus).

closed 2026-09-18 by bhall-ctre